YosysHQ/fpga-toolchain #4: Add ghdl, ghdl-yosys-plugin

edbordin, 2020-05-02:


daveshah1, 2020-05-02:
By following "Build as part of yosys (not recommended)" here: https://github.com/ghdl/ghdl-yosys-plugin I think we can get it to work on Windows, and as a fully static blob (the Yosys dynamic plugin interface doesn't support Windows yet).

edbordin, 2020-05-18:
After some patching I've managed to build it into yosys.  `GHDL_PREFIX` needs to be set on linux and darwin for it to work: `export GHDL_PREFIX=$install_dir/lib/ghdl`

[The Ada equivalent of argv[0]](https://github.com/ghdl/ghdl/blob/f01c93b00c618e994f795285a5eb3debeaf03fcf/src/ghdldrv/ghdllocal.adb#L330) is returning an empty string when ghdl is statically linked into yosys. For some reason it is working as intended on windows.

closed 2020-05-18 by edbordin

qwqw330, 2020-06-29:
Hey, if I try to run ```yosys -m ghdl ....``` I get the following error:

> ERROR: This version of yosys is built without plugin support.

In my .zshrc I have 
```
export PATH="$PATH:$HOME/ulx3s_workspace/fpga-toolchain/bin"
export GHDL_PREFIX="$HOME/ulx3s_workspace/fpga-toolchain/lib/ghdl"
```

Any help to fix this or do I have to compile yosys myself?

Thanks

edbordin, 2020-06-29:
Hi @qwqw330, as I understand it `-m ghdl` tells yosys to load a dynamic plugin for ghdl. Because we have compiled it into yosys you can simply remove that from the args and it should hopefully work. Let me know how it goes though because you are probably one of the first testing this build on an actual design!

qwqw330, 2020-06-29:
Oh that is very cool. Thank you, it works! Finally no more hourlong compiling. Yes! Thanks for your work.