YosysHQ/fpga-toolchain #7: Add verilator?

edbordin, 2020-05-02:


abhishek-kakkar, 2020-09-07:
I'm curious if there's a reason it's not there in these tools yet? I'd be interested in having these integrated into the fpga-toolchain package.

edbordin, 2020-09-07:
It hasn't been a priority because I figured if you are willing to compile your simulations then you are probably more comfortable building verilator from source. I haven't looked into the details but can imagine this tool would also not be easy to package up in a portable way - if it builds dynamic libs for your simulation to link against that would probably complicate matters. Also if it "learns" details of your system toolchain at configure time then that will not be portable to another system.

You're welcome to contribute something if you can find a way to make it work though.