YosysHQ/fpga-toolchain #25: Support for SymbiYosys

rajesh-s, 2020-06-03:
If possible, please include SymbiYosys (the formal verification tool, seeing increasing adoption by designers) in this package. Building/installing some of the solver dependencies is a pain.

edbordin, 2020-06-03:
Good suggestion! This has a lot of parts and it would probably deserve its own tarball release (ideally able to used standalone):

1. The frontend is a set of python scripts - currently this distribution doesn't bundle an interpreter so we would need to choose whether to bundle one or whether to try and use an external one (related: #19 ).

2. Building the various backends - I count five not including the yosys ones we already have https://symbiyosys.readthedocs.io/en/latest/quickstart.html#installing - Yices 2, Z3, super_prove, Avy, Boolector. This could be done in stages if certain engines are more popular.

Because it's a big job I can't say when I would be likely to start working on it, so I'm more than happy for someone else to take the lead on this. Either way I'll leave this open and it will probably happen at some point.

edbordin, 2020-09-01:
OK, after sitting on a branch for a while, SymbiYosys is now part of the package! I decided not to tackle the Avy and super_prover engines for now  - Avy is not supported on windows and I even ran into some issues building it for linux. super_prover runs on python2 using native code, so I think it would be more appropriate to package it for pip than to try and add it to this distribution.

closed 2020-09-01 by edbordin