YosysHQ/fpga-toolchain #37: yosys-smtbmc is broken in windows build

edbordin, 2020-07-05:
On other platforms this is a python script, for windows an exe launcher is generated but `make install` doesn't seem to be adding the script it expects.
```
$ yosys-smtbmc.exe --help
Cannot open <snip>\fpga-toolchain\bin\yosys-smtbmc-script.py
```

If I copy in the script from the linux build I get
```
$ yosys-smtbmc.exe --help
failed to create process.
```

https://github.com/YosysHQ/yosys/blob/93c6c6779861d7b41fae1b87d54790e0a9555b80/backends/smt2/smtbmc.py
https://github.com/YosysHQ/yosys/blob/93c6c6779861d7b41fae1b87d54790e0a9555b80/backends/smt2/Makefile.inc#L14
https://github.com/YosysHQ/yosys/blob/93c6c6779861d7b41fae1b87d54790e0a9555b80/misc/launcher.c

cr1901, 2020-07-06:
Discussed on IRC- the changes that broke @edbordin's setup were mine from a while back.

_By design_, the binary compiled from `launcher.c` will read the shebang line of `yosys-smtbmc-script.py`, and try to invoke a Python interpreter pointed to by that line. If the launcher can't find the interpreter, you get that helpful error.

The build system will attempt to search for a python binary and prepend `yosys-smtbmc-script.py` accordingly. However, if that search logic fails, a use can pass `make PYTHON=C:\path\to\python.exe` to force the shebang line to anything they want while `yosys` is installed.

edbordin, 2020-07-07:
fixed in #38 - worked out the shebang seems to work with a relative path so I bundled `python3.8.exe` but renamed it to `python3-private.exe`. We already have all of `lib/python3.8` for nextpnr anyway. Seems to work transparently and should avoid issues that might have otherwise been caused by putting MinGW python into `PATH`.

closed 2020-07-07 by edbordin