YosysHQ/fpga-toolchain #55: Z3 linked in the README is not a CAD software, but a SAT solver, a math software

yurivict, 2020-09-16:
The README says:
```
Z3: Engine for SymbiYosys
```

esden, 2020-09-16:
Not sure what you are trying to say. But Z3, the SAT solver is indeed the correct tool. SymbiYosys is a formal verification tool, thus it uses a SAT solver engine...

yurivict, 2020-09-16:
So if SymbiYosys uses Z3 then it should depend on Z3. Z3 in itself has nothing to do with FPGA toolchains. Z3 is just a solver.

esden, 2020-09-16:
You are wrong about it. Z3 is one of the many SAT solver engines that SymbiYosys supports. It is important to say which exact engines are included in the toolchain, and this is the intention of the README listing. Formal verification just like simulation is very much part of FPGA toolchain. I am closing this issue as I am convinced that the initial topic of this issue is "as intended". The SAT library link is the correct this case. Cheers.

closed 2020-09-16 by esden