YosysHQ/fpga-toolchain #57: Error while building avy (boost::logic::tribool conversion)

vmedea, 2020-10-06:
While building for `linux_x86_64` on Ubuntu 20.04 Focal, the following error appears and stops the build:
```
In file included from /home/ubuntu/fpga-toolchain/_builds/build_linux_x86_64/avy/avy/src/ItpMinisat.cc:2:
/home/ubuntu/fpga-toolchain/_builds/build_linux_x86_64/avy/avy/src/ItpMinisat.h: In member function 'bool avy::ItpMinisat::isSolved()':
/home/ubuntu/fpga-toolchain/_builds/build_linux_x86_64/avy/avy/src/ItpMinisat.h:127:52: error: cannot convert 'boost::logic::tribool' to 'bool' in return
  127 |     bool isSolved () { return m_Trivial || m_State || !m_State; }
      |                               ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
      |                                                    |
      |                                                    boost::logic::tribool
[ 98%] Built target minisat_core.BIN
[ 98%] Linking CXX executable minisat
[ 98%] Linking CXX executable glucose
[ 98%] Built target minisat
In file included from /home/ubuntu/fpga-toolchain/_builds/build_linux_x86_64/avy/avy/src/ItpGlucose.cc:2:
/home/ubuntu/fpga-toolchain/_builds/build_linux_x86_64/avy/avy/src/ItpGlucose.h: In member function 'bool avy::ItpGlucose::isSolved()':
/home/ubuntu/fpga-toolchain/_builds/build_linux_x86_64/avy/avy/src/ItpGlucose.h:129:52: error: cannot convert 'boost::logic::tribool' to 'bool' in return
  129 |     bool isSolved () { return m_Trivial || m_State || !m_State; }
      |                               ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
      |                                                    |
      |                                                    boost::logic::tribool
/home/ubuntu/fpga-toolchain/_builds/build_linux_x86_64/avy/avy/src/ItpGlucose.h: In member function 'bool avy::ItpGlucose::getVarVal(int)':
/home/ubuntu/fpga-toolchain/_builds/build_linux_x86_64/avy/avy/src/ItpGlucose.h:185:23: error: cannot convert 'boost::logic::tribool' to 'bool' in return
  185 |         return tobool (m_pSat->modelValue(x));
      |                ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
      |                       |
      |                       boost::logic::tribool
```
I suspect it is a boost version incompatibility, though I'm not sure the version of boost in this version of ubuntu is too old or too new.

edbordin, 2020-10-08:
Yeah, I left Avy disabled after hitting issues like this. I think there were even more issues trying to build for OS X and Windows...

likewise, 2021-07-30:
*UNTESTED* - Based on a similar patch: https://svnweb.freebsd.org/ports/head/multimedia/mkvtoolnix/files/patch-boost-1.69?view=markup&pathrev=482787

```
index 657253d..66a0cd4 100644
--- a/src/ItpGlucose.h
+++ b/src/ItpGlucose.h
@@ -126,7 +126,7 @@ namespace avy
     ::Glucose::Solver* get () { return m_pSat; }
     
     /// true if the context is decided 
-    bool isSolved () { return m_Trivial || m_State || !m_State; }
+    bool isSolved () { return bool{m_Trivial || m_State || !m_State}; }
 
     int core (int **out)
     {
@@ -182,7 +182,8 @@ namespace avy
     bool getVarVal(int v)
     {
         ::Glucose::Var x = v;
-        return tobool (m_pSat->modelValue(x));
+        boost::logic::tribool y = tobool (m_pSat->modelValue(x));
+        return bool{y};
     }
   };
   
diff --git a/src/ItpMinisat.h b/src/ItpMinisat.h
index d145d7c..7514f31 100644
--- a/src/ItpMinisat.h
+++ b/src/ItpMinisat.h
@@ -124,7 +124,7 @@ namespace avy
     ::Minisat::Solver* get () { return m_pSat.get (); }
     
     /// true if the context is decided 
-    bool isSolved () { return m_Trivial || m_State || !m_State; }
+    bool isSolved () { return bool{m_Trivial || m_State || !m_State}; }
 
     int core (int **out)
     {
```

likewise, 2021-09-24:
Thank you for the update. Your repo is very valuable still!  With the above patch I could integrate Avy at least under Linux and if people find the same issue it might work for them also.

closed 2021-09-24 by vmedea