YosysHQ/fpga-toolchain #68: Nextpnr GUI support

MartinGeisse, 2021-01-01:
Would it be possible to compile nextpnr, in particular nextpnr-ecp5, with GUI support?

According to https://github.com/YosysHQ/nextpnr GUI support is disabled by default to reduce dependencies. It is not immediately clear to me whether this is related to build-time dependencies or run-time dependencies. At run-time, the GUI must be enabled with a command-line option, but again it is not clear to me whether one can run nextpnr headless on a headless system just by ommitting that option once it was built *with* GUI support.

All this could imply that GUI support can only be provided by building two separate executables, nextpnr-ecp5 and nextpnr-ecp5-gui, for headless and gui mode, respectively.

With respect to ccompile-time dependencies, the manual says "On Ubuntu, install qt5-default", but this sounds like it would install the qt5 runtime on the build host, not the necessary compile-time files to build a qt5 executable that is intended to be run on a different machine. In other words, it seems to confuse the build host and the target machine.

edbordin, 2021-01-02:
It's mainly left off at the moment because it was assumed that statically linking QT into two nextpnr binaries will add a significant amount to the overall package size. But I never actually checked how much so I'd be open to at least trying it and quantifying the size difference

ArsenioDev, 2021-02-02:
Please do, I quite like the UI and would like it in these bins

edbordin, 2021-09-24:
I decided months ago not to make any more changes to this package, I should have done a better job of communicating that, sorry! I've added a notice to the README.