YosysHQ/fpga-toolchain #73: Adding Project Apicula

ArsenioDev, 2021-02-02:
I'm looking to spool up teaching and turnkey starter kits of getting your feet wet in FPGA and my board of choice is the Sipeed Tang Nano, supported by the Apicula interface layer. 
If you could add build of the stack into this repo that'd be greatly appreciated!

edbordin, 2021-02-02:
It probably depends whether @pepijndevos thinks it's mature enough to start shipping it. Is it merged into upstream nextpnr yet?

pepijndevos, 2021-02-02:
It's upstream in nextpnr, yes. It works for simple designs, but is not quite ready for large designs as it lacks things like BRAM, PLL, Serdes, and other advanced features. So if the purpose is for people to do their first blinkies without having to sign up for a license, it might work actually? If some organization wants to sponsor the development of certain features this is of course possible to discuss ;)

One thing that's one the roadmap is to get Apicula into Yowasp. Once this is done you could set up the entire toolchain with a single pip install.

Apicula itself is a Python package, so making "binaries" for that only makes sense if you want to bundle Python for example on Windows.  Adding the nextpnr target should be straightforward though.

edbordin, 2021-09-24:
I decided months ago not to make any more changes to this package, I should have done a better job of communicating that, sorry! I've added a notice to the README.