YosysHQ/fpga-toolchain #76: Mac GHDL binary references paths from build environment

controversial, 2021-03-08:
On the latest nightly build, `ghdl-yosys-plugin` references paths that only exist in the build environment:

```
$ ./bin/yosys -p ghdl

-- Running command `ghdl' --

1. Executing GHDL.
warning: ieee library directory '/Users/runner/work/1/s/_packages/build_darwin/fpga-toolchain/lib/ghdl/ieee/v93/' not found
error: cannot find "std" library

raised OPTIONS.OPTION_ERROR : ghdlcomp.adb:315
```

Ideally, the binary would be built to be portable and run in different environments

edbordin, 2021-03-08:
This is also the case with the Linux build, see [the README](https://github.com/YosysHQ/fpga-toolchain#using-ghdl) for how to work around this with the `GHDL_PREFIX` environment variable.

There was some discussion [here](https://github.com/YosysHQ/fpga-toolchain/pull/59) on ways we could change the absolute prefix or make the plugin search relative to the install location as it does on windows. (cc: @tgingold @umarcor )

controversial, 2021-03-09:
Thanks, setting GHDL_PREFIX fixes it; I hadn't seen that. Feel free to close this or to leave it open as an issue to track the more robust solution you described

closed 2021-09-24 by edbordin