YosysHQ/fpga-toolchain #78: No new releases for Windows

pcossutta, 2021-04-30:
The last release for windows is old, are you dropping support for Windows or is something that's not working ?
By the way, thanks for this amazing job you are doing !

edbordin, 2021-05-17:
I tried to have a look a while ago and concluded it wasn't a quick fix. I think I need to turn off link-time-optimization in the nextpnr and yosys builds for it to work under msys2's mingw64 toolchain, but I probably need to create PRs upstream in both projects to make that an option (or set up more patches). I'll prioritise #79 but then I'll try to fix this.

edbordin, 2021-09-24:
I decided months ago not to make any more changes to this package, I should have done a better job of communicating that, sorry! I've added a notice to the README.