YosysHQ/fpga-toolchain #82: readline support ...

a2800276, 2021-08-10:
Hope I'm not beating a dead horse:  how come the bundled yosys is built without readline support? I initally assumed I was probably doing something stupid wrong, but from the looks of it, readline support is actively disabled in: [`compile_yosys.sh`](https://github.com/YosysHQ/fpga-toolchain/blob/5d78f2d9f5b64b6f08ab3c90a866f9d8f75f8ecc/scripts/compile_yosys.sh#L86)

I was wondering if there is a reason for this? Else I would be happy to set up some patches for downloading deps and building with support enabled. This distribution of tools is incredibly helpful, but not having tab-completion is a major annoyance (for me :-)   ...

edbordin, 2021-09-24:
I decided months ago not to make any more changes to this package, I should have done a better job of communicating that, sorry! I've added a notice to the README.

a2800276, 2021-09-27:
I have realized this in the meantime, thanks for updating the README.

closed 2021-09-27 by a2800276